platformio/platform-espressif8266 #224: Build errors xtensa-lx106-elf-gcc: error: unrecognized command line option '-std=c17'

TD-er, 2020-07-23:
Latest code base of esp8266/Arduino cannot be built anymore.
Apparently we're using the wrong compiler.

What can be done to be able to compile the latest "master" branch of esp8266/Arduino?

mcspr, 2020-07-23:
ref. https://github.com/esp8266/Arduino/pull/6294#issuecomment-640840956
just add another package line with `toolchain-xtensa @ $URL`, where $URL is from here https://github.com/earlephilhower/esp-quick-toolchain/releases/tag/3.0.0-gnu12

Current installation instructions expect Arduino IDE:
https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version
By manually running tools/get.py, which downloads latest toolchain+utils based on .json package lists maintained by the Core authors. But, not sure how to get that information into the platformio.ini automatically when using git version

TD-er, 2020-07-24:
OK, I got 1 step further, but apparently it is not enough.

In my platformio.ini:
```
[core_stage]
extends                   = esp82xx_2_6_x
platform                  = https://github.com/platformio/platform-espressif8266.git
build_flags               = ${esp82xx_2_6_x.build_flags}
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_191122
                            -Wno-deprecated-declarations
platform_packages         =
  framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
  toolchain-xtensa @ https://github.com/earlephilhower/esp-quick-toolchain.git#3.0.0-gnu12
```

So it does install something...

```
Processing test_beta_ESP8266_4M1M (platform: https://github.com/platformio/platform-espressif8266.git; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------PackageManager: Installing toolchain-xtensa
git version 2.21.0.windows.1
Cloning into 'C:\Users\gijs\.platformio\packages\_tmp_installing-56u6vfax-package'...
remote: Enumerating objects: 66, done.
remote: Counting objects: 100% (66/66), done.
remote: Compressing objects: 100% (57/57), done.

Unpacking objects: 100% (66/66), done.
Note: checking out '0474ae9ad28a3d1e872b5b68eee54761441b963c'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

toolchain-xtensa @ 0474ae9 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
```

Build error:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 2.6.0 #b27329b > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 1019.98KB Flash
PACKAGES: 
 - framework-arduinoespressif8266 83fc47f
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 0474ae9
Converting ESPEasy.ino
'xtensa-lx106-elf-g++' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1
AssertionError: :
  File "C:\Users\gijs\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 169:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\gijs\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\gijs\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\gijs\.platformio\platforms\espressif8266@src-eb7495f88eb0afa18fedff98bfb5e40f\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\gijs\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\gijs\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\gijs\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\gijs\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 118:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\gijs\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\gijs\.platformio\penv\Lib\site-packages\platformio\builder\tools\platformio.py", line 330:
    env.ConvertInoToCpp()
  File "C:\Users\gijs\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\gijs\.platformio\penv\Lib\site-packages\platformio\builder\tools\piomisc.py", line 239:
    out_file = c.convert(ino_nodes)
  File "C:\Users\gijs\.platformio\penv\Lib\site-packages\platformio\builder\tools\piomisc.py", line 91:
    return self.process(contents)
  File "C:\Users\gijs\.platformio\penv\Lib\site-packages\platformio\builder\tools\piomisc.py", line 112:
    assert self._gcc_preprocess(contents, out_file)
===================================================================================================================================================== [FAILED] Took 8.53 seconds ===================================================================================================================================================== 
```

mcspr, 2020-07-24:
You need to use the OS-specific .zip / tarball from releases, see my link:
https://github.com/xoseperez/espurna/blob/ea111c4cf2229502c19fc739737527670c20ec1e/code/platformio.ini#L124
> toolchain-xtensa @ https://github.com/earlephilhower/esp-quick-toolchain/releases/download/3.0.0-gnu12/x86_64-linux-gnu.xtensa-lx106-elf-0474ae9.200706.tar.gz

TD-er, 2020-07-24:
But that's not usable.
I cannot make a .ini file for every OS we need to build on, let alone for every platform my users are building on.

mcspr, 2020-07-24:
Like I said, list of toolchains is provided in the platform (`platform = espressif8266`). For example, this was done for Tasmota:
https://github.com/Jason2866/platform-espressif8266/tree/new_gcc
https://github.com/Jason2866/platform-espressif8266/blob/new_gcc/platform.json
https://github.com/Jason2866/platform-espressif8266/releases/download/3.0.0-gnu12/manifest.json
Modification consists of updated platform.json that includes a custom 'manifest.json' with a gcc10.1 os<->url list and use a different (`>=5`) toolchain version.

This list is in the Core repo:
https://github.com/esp8266/Arduino/blob/83fc47f6d97d6a82acb6cf2c856381bcd101f0c0/package/package_esp8266com_index.template.json#L201-L255

platform_packages = ... simply overrides the package from the package.json by name, but I am not sure if it is modifiable via scriptin.
https://github.com/platformio/platformio-core/blob/85f5a6a84a6680733ab2baeb9eeb8f08be110f49/platformio/managers/platform.py#L713
https://github.com/platformio/platformio-core/blob/85f5a6a84a6680733ab2baeb9eeb8f08be110f49/platformio/managers/platform.py#L637-L645

What I was suggesting is to find a solution of integrating it with PIO (...moving manifest & platform to the esp8266/Arduino repo?), avoiding such weird configuration where you end up the maintainer of the package list.

TD-er, 2020-07-24:
Well at least I now have it up and running on my Windows environment.
So I'm already working on the build issues when compiling with gcc10.1 
I really hope the PlatformIO developers can help out here to what could be done to make it fool-proof (... said by the fool)

mcspr, 2020-07-26:
True, but hopefully this is something temporary.
afaik, what I proposed above with config scripting would not work, b/c when esp8266 Core's tools/platformio-build.py is called, package was already downloaded. Same with extra scripts, because those are loaded too late, too.

I wonder if platformio could try to read a specific file inside of the selected "framework" package, which will list the additional package files; i.e., adding itself to the in platform.json's "packageRepositories" list. Or, add platform_repositories = ... URLs in addition to the existing platform_packages = ..., so we could add those directly via platformio.ini
Then, .ini would only need to specify the version number and OS will be chosen automatically.

But, this becomes an issue for the https://github.com/platformio/platformio-core/issues

valeros, 2020-07-31:
Hi guys, I've just uploaded gcc v10.1.0 to our CDN, so you should be able to override the default toolchain in your platformio.ini:
```ini
[env:nodemcuv2]
platform = espressif8266
framework = arduino
board = nodemcuv2
monitor_speed = 115200
platform_packages = 
    toolchain-xtensa @ ~2.100100.0
```
Please let me know if it works, thanks!

TD-er, 2020-07-31:
Apart from the build issues with GCC10.1 I still need to fix, it both installs on Windows 10 (64 bit) and the Ubuntu20.04 I run on Linux for Windows subsystem.
So the toolchain issue seems to work.

valeros, 2020-07-31:
Is there anything else I can help you with?

TD-er, 2020-07-31:
Not right now, so unless someone else finds an environment where the toolchain cannot be installed, I would say this issue can be closed.

closed 2020-07-31 by valeros

mcspr, 2020-07-31:
When upstream changes to a newer version (e.g. 10.2) while still in development, will it somehow be tracked and uploaded to the PIO registry? By watching https://github.com/earlephilhower/esp-quick-toolchain/releases automatically or otherwise.

btw any thoughts on the `platform_repositories = ... ` override?

valeros, 2020-07-31:
Hi @mcspr 

> When upstream changes to a newer version (e.g. 10.2) while still in development, will it somehow be tracked and uploaded to the PIO registry?

Unfortunately, there is no easy way to achieve that. Only stable/critical updates are uploaded to PlatformIO CDN.

> btw any thoughts on the platform_repositories = ... override?

It might be a good idea to open a feature request in the `platformio-core` repository.

Jason2866, 2020-07-31:
@mcspr  since earlephilhower added a package.json you can easily build your own.
I have done (before this) for using gcc10.1 build chain with platformio.
See https://github.com/Jason2866/platform-espressif8266/releases/tag/3.0.0-gnu12

mcspr, 2020-08-01:
@Jason2866 Yes, I had noticed that! My only issue is besides the manifest, we also drag python code included in the platform when we fork it. The idea here is to allow to override the manifest.json through .ini directly, while still using the original `platform = espressif8266`

Meaning, something like:
```ini
[env:test]
platform = espressif8266
framework = arduino
board = d1_mini
platform_repositories =
    my_own_manifest_with_toolchain_xtensa.json ; e.g. esp8266/Arduino CI could ship / convert existing ArduinoIDE files without much hassle
platform_packages = 
    framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
    toolchain-xtensa @ ~2.100200.0 ; future 10.2
```

Jason2866, 2020-08-01:
@mcspr  yes, that would  be a nice feature.

arihantdaga, 2020-09-09:
After updating similar to https://github.com/xoseperez/espurna/commit/8510f77d1262ddb971f8e2300b8e2dd51fd9c8ba in my project i am getting many errors including declaration of byte. 
```
/home/arihant/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:241:27: error: reference to 'byte' is ambiguous
  241 | uint16_t makeWord(byte h, byte l);

AES/printf.h:31:3: error: 'fdevopen' was not declared in this scope; did you mean 'fmemopen'?

NtpClientLib/src/NtpClientLib.h:427:44: error: 'byte' has not been declared
```

AND, It is happening only when i am using platfomio, with Arduino It is compiling without any problem. 
My configuration looks like this - 
```
[env:test]
platform = espressif8266
platform_packages = 
        framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#2843a5a
        toolchain-xtensa @ ~2.100100.200706

framework = arduino
board = nodemcuv2
upload_speed = 115200
```
* The Arduino core is art git version 2.7.4

mcspr, 2020-09-09:
@arihantdaga as a random guess - do you have `using namespace std;` somewhere in libs / project? (before Arduino.h include, for example)
This will inadvertently make  `std::byte` into `byte`, which **will** conflict with the thing in the Arduino.h (see `typedef uint8_t byte;` above `makeWord()`)