platformio/platform-espressif8266 #225: mklittlefs not found when running inside a github actions workflow

rgl, 2020-07-27:
What kind of issue is this?

- [X] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

**NB** I'm sorry, but I have to ask a question here, because I never receive the email sent by https://community.platformio.org/ while trying to logging in with GitHub.

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 18.04

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 4.3.4

### Description of problem

mklittlefs not found when running inside a github actions workflow:

https://github.com/rgl/wemos-d1-mini-esp8266-ds3231-rtc-module/runs/908155922?check_suite_focus=true#step:11:56

there is something odd happening because it tries to install the tool-mkspiffs package instead of installing the tool-mklittlefs one.

it works fine locally in my machine, but for some reason it does not work when running in a GitHub action. do you have any idea why?

#### Steps to Reproduce

1. fork the code at https://github.com/rgl/wemos-d1-mini-esp8266-ds3231-rtc-module/blob/add-web-ui/
2. trigger the github action in your fork (in the add-web-ui branch)

### Actual Results

```
+ platformio run --target buildfs
Downloading...
Unpacking...
tool-mkspiffs @ 1.200.0 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.6.0 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20702.0 (2.7.2) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - tool-mkspiffs 1.200.0 (2.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing id=83 @ 1.10.0
Using cache: /home/runner/.platformio/.cache/8e/dae588c6aa803a03e14a87f287a1c58e
Unpacking...
RTClib @ 1.10.0 has been successfully installed!
Installing dependencies
Looking for TinyWireM library in registry
Found: https://platformio.org/lib/show/797/TinyWireM
LibraryManager: Installing id=797
Using cache: /home/runner/.platformio/.cache/5f/c601e442f4577465905ff1d0e3a6bd5f
Unpacking...
TinyWireM @ 1.1.0 has been successfully installed!
LibraryManager: Installing id=64 @ 6.15.2
Using cache: /home/runner/.platformio/.cache/7e/7acaf4b435abc12fc6eceb805cca3e7e
Unpacking...
ArduinoJson @ 6.15.2 has been successfully installed!
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ArduinoJson> 6.15.2
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <LittleFS(esp8266)> 0.1.0
|-- <RTClib> 1.10.0
|   |-- <Wire> 1.0
|-- <Wire> 1.0
Building in release mode
Building file system image from 'data' directory to .pio/build/d1_mini/littlefs.bin
sh: 1: mklittlefs: not found
*** [.pio/build/d1_mini/littlefs.bin] Error 127
```

### Expected Results

I expected it to create littlefs.bin file.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**

See it at https://github.com/rgl/wemos-d1-mini-esp8266-ds3231-rtc-module/blob/add-web-ui/platformio.ini

valeros, 2020-07-30:
Resolved in the latest release, please update.

closed 2020-07-30 by valeros

rgl, 2020-07-30:
Thank you!

I'm now worried that my platformio project is not locking/fixing its dependencies, as it automatically installed this new version, going to look into fixing that :-)