platformio/platform-espressif8266 #226: ESP8266 arduino framework and loop()

hgschmie, 2020-08-17:
Not sure that this is the right place to ask.

I use platformio 4.3.4 with the arduino framework. In various sketches that I looked at, it is implied that `loop()` is called within an endless loop so exiting the function is not a big deal, it will be called again. I even found posts around esp8266 and loop implying that you can not run too much functionality within a single loop as this may affect WiFi and other functionality. 

*BUT* in the platformio version of Arduino.h (from framework-arduinoespressif8266  @ 3.20704.0), the `loop()` function is marked as `[[noreturn]]` and lo and behold, if my code actually exits loop(), the board crashes with a stack trace. 

This is even more confusing, as in the official arduino repo (https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Arduino.h#L192), it is not. The platformio framework claims that it is based on 2.7.4 (https://github.com/platformio/platform-espressif8266/commit/fbbe4e2135920a82ae4ca7750f10de155cd84bc9).

Diffing the cores/esp8266 tree from this repo with the locally installed ~/.platformio/packages/framework-arduinoespressif8266/cores/esp8266 literally shows this as the only significant difference between the arduino repo and what I see locally:

```
diff -urb Arduino/cores/esp8266/Arduino.h /home/henning/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h
--- Arduino/cores/esp8266/Arduino.h	2020-08-17 14:09:18.245135688 -0700
+++ /home/henning/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h	2020-08-14 09:47:57.771492095 -0700
@@ -201,7 +201,7 @@
 
 void preinit(void);
 void setup(void);
-void loop(void);
+[[noreturn]] void loop(void);
 
 void yield(void);
 
diff -urb Arduino/cores/esp8266/core_esp8266_main.cpp /home/henning/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp
--- Arduino/cores/esp8266/core_esp8266_main.cpp	2020-08-17 14:09:18.246135675 -0700
+++ /home/henning/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp	2020-08-14 09:47:57.786491890 -0700
@@ -40,7 +40,8 @@
 #define LOOP_QUEUE_SIZE    1
 
 extern "C" void call_user_start();
-extern void loop();
+
+[[noreturn]] extern void loop();
 extern void setup();
 extern void (*__init_array_start)(void);
 extern void (*__init_array_end)(void);
```

So my guess is that there is a reason for this. But I don't understand it. This seems to be platformio specific and it requires me to rewrite code that works on non-platformio arduino code. What is more confusing, I don't seem to be able to pinpoint the platce where this change is made nor am I able to find documentation as to why this is changed in platformio.

valeros, 2020-08-18:
Hi @hgschmie ! To be honest I have no idea how how that ``[[noreturn]]`` attribute appeared in your `framework-arduinoespressif8266`, I checked our package here https://bintray.com/platformio/dl-packages/framework-arduinoespressif8266/3.20704.0#files and there is no difference with the original Arduino framework. Are you sure all your packages are up-to-date?

hgschmie, 2020-08-18:
I figured it out. It is an IDE bug: https://youtrack.jetbrains.com/issue/CPP-21848

closed 2020-08-18 by hgschmie