platformio/platform-espressif8266 #227: Project working on Arduino IDE fails to build in PIO

pfeerick, 2020-08-31:
In the process of porting a project written by someone else from Arduino IDE to PlatformIO, and hit a snag with 

`UdpContext.h:539:34: error: 'DEBUGV' was not declared in this scope`

The project is on GitLab (platformio branch), and I set up a runner to do CI builds for this, which gives the same error as my own desktop (on Windows). Latest version of the Arduino IDE with the same platform and library versions (i.e. all latest versions) results in the same erorr. Runner log and the Arduino IDE verbose log are attached in the issue linked to the CI failure: https://gitlab.com/pfeerick/mrdiy-audio-notifier/-/issues/1 . As I mention in that issue, adding

```
#ifndef DEBUGV
#define DEBUGV(...) do { (void)0; } while (0)
#endif
```

to a framework core file - `%userprofile%\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src\include\UdpContext.h` - resolves it... so is it something I've missed, or is there a PIO incompatibility here?

max5555, 2020-10-11:
I have the same problem and your solution helped.

pfeerick, 2020-10-16:
Thanks for the feedback. Was it for the same project - _mrdiy-audio-notifier_? As if it isn't, it would be interesting to know what it is, so we can see where else it has failed. I haven't tried to compile this again since, so I don't know if anything else has changed in the meantime.

EtienneTetreault, 2020-11-05:
Same problem here and the solution (define `DEBUGV` in UdpContext.h) seems to works. I did use it for the mrdiy-audio-notifier project. On another note, for the same project, I had to declare #include `#include <SD.h>` in the main.cpp to ensure the ESP8266Audio librairy would find it. I dont think it is related though.

pfeerick, 2020-11-23:
In order to work around this for now, since I'm revisiting the project again, I used an `extra_script` to apply this workaround if not already applied. You can see the what I did [here](https://gitlab.com/pfeerick/mrdiy-audio-notifier/-/commit/3a04e4ab913ca29bf05098b05ffe92e0ed8c7e9a). It's nothing fancy, and is basically a copy of the [how to override a package file example from the docs](https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#override-package-files), and seems to work fine for Windows and Linux (CI linux runner was fine, and on Windows I have git installed).