platformio/platform-espressif8266 #228: Error: Could not find LD script when uploading remotely

mariusstaicu, 2020-09-01:
PIO version 4.3.4
for every espressif8266 platform version > 2.5.1 I get this error when I'm trying to upload firmware remotely - 
```
pio remote -a agent run -e env -t upload --upload-port 192.168.1.141 
```
```
Uploading firmware remotely
Processing release (platform: espressif8266@2.6.2; board: nodemcuv2; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Could not find LD script
========================== [FAILED] Took 3.45 seconds ==========================

Environment    Status    Duration
-------------  --------  ------------
development    IGNORED
integration    IGNORED
release        FAILED    00:00:03.451
production     IGNORED
==================== 1 failed, 0 succeeded in 00:00:03.451 ====================
```

espressif8266@2.5.1 is the latest one that works correctly.

valeros, 2020-09-03:
Hi @mariusstaicu ! Does it work if you specify `--force-remote` flag?

mariusstaicu, 2020-09-03:
using force-remote seems to be working, however it is not good for my case, I need the build to happen on local system

mariusstaicu, 2020-12-11:
any updates on this ?

vvf, 2021-07-21:
local PIO version 5.1.1
remote PIO version 5.1.1

`> Executing task in folder esp12e-relay8: platformio remote run --target upload <

Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (2.6.3) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-mklittlefs 1.203.210203 (2.3) 
 - tool-mkspiffs 1.200.0 (2.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Linking .pio/build/esp12e/firmware.elf
/home/vvf/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/esp12e/libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
/home/vvf/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/esp12e/libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
/home/vvf/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/esp12e/libFrameworkArduino.a(core_esp8266_main.cpp.o): in function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
/home/vvf/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp12e/firmware.elf] Error 1
========================== [FAILED] Took 1.64 seconds ==========================
The terminal process "platformio 'remote', 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
`

Still not worked!
When build local - found all  dependencies and compile. 
I tried to cleanup and than remote upload - all the same.