platformio/platform-espressif8266 #230: Can't install tool-esptool on aarch64 system

MatejSpindler, 2020-09-25:
### Configuration

**Operating system**: Manjaro ARM linux on PinebookPro (aarch64)

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.3.4

### Description of problem
During installation of tool-esptools it looks like PackageManager tries to install armv6l version instead of aarch64.

#### Steps to Reproduce
```
(test) matej@PinebookPro ~/tt> rm -rf /home/matej/.platformio
(test) matej@PinebookPro ~/tt> pio run
*****************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
****************************************************************************************************

Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
----------------------------------------------------------------------------------------------------
PlatformManager: Installing espressif8266
espressif8266 @ 2.6.2 has been successfully installed!
The platform 'espressif8266' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-xtensa @ ~2.40802.191122
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
toolchain-xtensa @ 2.40802.200502 has been successfully installed!
PackageManager: Installing framework-arduinoespressif8266 @ ~3.20704.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
framework-arduinoespressif8266 @ 3.20704.0 has been successfully installed!
PackageManager: Installing tool-esptool @ <2
Warning! Package Mirror: The 'sha1' sum 'e15fba5accf70d9c1574c47055ee9aedc0c3fe1c' of downloaded file 'tool-esptool-linux_armv6l-1.413.0.tar.gz' is not equal to remote 'bdfd4890e232df1876df63c8e676b12119e89c9f'
Looking for another mirror...
Error: Could not install 'tool-esptool' with version requirements '<2' for your system 'linux_aarch64'.

Please try this solution -> http://bit.ly/faq-package-manager
```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() { }; 
void loop() { };
```

### Additional info
With manually downloaded package `tool-esptool-linux_aarch64-1.413.0.tar.gz` from SourceForge and extracted  to `$HOME/.platformio/packages/` firmware can be compiled and UPLOADED to the target.

ACrazyConcept, 2020-12-30:
I have the same issue. Except it is just for the arm versions:

RPi 4:
```
Processing shp6 (board: esp8285; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing tool-esptool @ <2
Warning! Package Mirror: The 'sha1' sum 'e15fba5accf70d9c1574c47055ee9aedc0c3fe1c' of downloaded file 'tool-esptool-linux_armv6l-1.413.0.tar.gz' is not equal to remote 'bdfd4890e232df1876df63c8e676b12119e89c9f'
Looking for another mirror...
Error: Could not install 'tool-esptool' with version requirements '<2' for your system 'linux_armv7l'.

Please try this solution -> http://bit.ly/faq-package-manager
```

RPi Zero W:
```
Processing shp6 (board: esp8285; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing tool-esptool @ <2
Warning! Package Mirror: The 'sha1' sum 'e15fba5accf70d9c1574c47055ee9aedc0c3fe1c' of downloaded file 'tool-esptool-linux_armv6l-1.413.0.tar.gz' is not equal to remote 'bdfd4890e232df1876df63c8e676b12119e89c9f'
Looking for another mirror...
Error: Could not install 'tool-esptool' with version requirements '<2' for your system 'linux_armv6l'.

Please try this solution -> http://bit.ly/faq-package-manager
```

keesj, 2021-01-02:
After performing pio upgrade and platformio upgrade the problem disappeared

ACrazyConcept, 2021-01-12:
> 
> 
> After performing pio upgrade and platformio upgrade the problem disappeared

I concur.

keesj, 2021-01-16:
The thing is that I really recently installed the package as dependency of esphome hence did not expect this problem.

valeros, 2021-02-03:
Could someone verify that the problem still persists after updating everything to the latest versions?

MatejSpindler, 2021-02-03:
> Could someone verify that the problem still persists after updating everything to the latest versions?

In latest version this issue is fixed.

closed 2021-02-03 by MatejSpindler