platformio/platform-espressif8266 #232: MacOS Big Sur esptoolpy ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found

maxgerhardt, 2020-11-14:
Tracking bug report per https://community.platformio.org/t/esp8266-big-sur-11-0-1/17129?u=maxgerhardt. It seems like the Big Sur 11.0.1 update makes the pyserial module in `esptool.py` misbehave. 

```
Traceback (most recent call last): 
File "/Users/pc/.platformio/packages/tool-esptoolpy/esptool.py", line 57, in <module> 
import serial.tools.list_ports as list_ports 
File "/Users/pc/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports.py", line 29, in <module> from serial.tools.list_ports_posix 
import comports 
File "/Users/pc/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports_posix.py", line 31, in <module> from serial.tools.list_ports_osx 
import comports 
File "/Users/pc/.platformio/penv/lib/python3.8/site-packages/serial/tools/list_ports_osx.py", line 32, in <module> 
kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault") 
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
```

valeros, 2020-11-19:
Thanks!

esptoolpy v3.0 will be used by default in the next platform release. As for now users on Big Sur can override the package directly in their configuration:
```
platform_packages =
    platformio/tool-esptoolpy @ ~1.30000.0
```

closed 2020-11-19 by valeros

wiscot, 2020-11-19:
> Thanks!
> 
> esptoolpy v3.0 will be used by default in the next platform release. As for now users on Big Sur can override the package directly in their configuration:
> 
> ```
> platform_packages =
>     platformio/tool-esptoolpy @ ~1.30000.0
> ```

Dear valeros, I have the same issue than maxgerhardt and with this trick he works fine.

Thank you very much!

commit 81912f3b (2020-11-19): Update esptoolpy to v3.0 // Resolve #232