platformio/platform-espressif8266 #238: Add project task to download configured filesystem from MCU again

maxgerhardt, 2020-12-31:
We do have the `uploadfs` target to pack and upload our `data/` folder as a filesystem (SPIFFS or LittleFS) binary to the ESP8266, but there is no built-in reverse: To download the memory contents of flash at the right place and unpack the filesystem again.

This feature request was also made in https://community.platformio.org/t/download-or-view-esp8266-file-system/17749 and https://community.platformio.org/t/esp8266fs-file-uploader-for-esp/16327 and https://community.platformio.org/t/littlefs-how-to-dump-download-the-file-system-image-from-board-to-computer/18044, so there is some demand for it. 

I've created an `extra_scripts` extension at https://github.com/maxgerhardt/pio-esp32-esp8266-filesystem-downloader that works for both ESP8266 and ESP32. By duplicating some of the platform's code (which I somehow couldn't `import`.. would give a weird base64 error..), I can reconstruct where the filesystem is supposed to be, download it and decode it using `mklittlefs` or `mkspiffs` respectively. 

It would be great if this code (or a better version of it) would be directly in the platform (ESP8266 + ESP32) so that it's a built-in project task to download the configured filesystem again.

Jason2866, 2021-01-13:
@maxgerhardt  Couldnt believe that the function is not there!
Found this Feature Request by searching for `how to download the content of the filesystem with platformio`.
Since we use littlefs already with ESP32 in Project [Tasmota](https://github.com/arendst/Tasmota) it would be really great if you add the littlefs part for ESP32.
Thx!

maxgerhardt, 2021-01-13:
Indeed! Note that there's a fork https://github.com/maxromanovsky/pio-esp32-esp8266-filesystem-downloader which fixes some command-invocation errors. On a Mac OS environment, the script is [reported as non-working](https://community.platformio.org/t/download-or-view-esp8266-file-system/17749/7?u=maxgerhardt) though (downloaded content is not decodable), but works on my ESP8266 board and my Windows machine, so the script might need more work after all..

PRs are very welcome to enhance the script and maybe even detect that the `LittleFS_esp32` library is used to switch to the LittleFS decoder :)