platformio/platform-espressif8266 #240: Exception 29: StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores

undermuz, 2021-01-25:
Every 20-40 seconds my nodemcuv2 crash with this error:
```
PC: 0x4020b599
EXCVADDR: 0x00000000

Decoding stack results
0x4023a159: tcp_input at core/tcp_in.c line 942
0x4023ff22: ip4_input at core/ipv4/ip4.c line 1467
0x40236f29: ethernet_input_LWIP2 at netif/ethernet.c line 188
0x40236d48: esp2glue_ethernet_input at glue-lwip/lwip-git.c line 469
0x4025c6b2: ethernet_input at glue-esp/lwip-esp.c line 365
0x4025c6c3: ethernet_input at glue-esp/lwip-esp.c line 373
```

valeros, 2021-01-25:
Hi @undermuz ! Could you try to compile your project using Arduino IDE? Does it work there?

undermuz, 2021-01-30:
> Hi @undermuz ! Could you try to compile your project using Arduino IDE? Does it work there?

@valeros Hi! How can i do that?