platformio/platform-espressif8266 #242: Update toolchain-xtensa GCC 10.2.0 version

maxgerhardt, 2021-01-30:
The current `toolchain-xtensa` version (https://bintray.com/platformio/tool-packages/toolchain-xtensa-windows_amd64#files e.g.) is from the second-latest release at https://github.com/earlephilhower/esp-quick-toolchain/releases, GCC 10.1.0 compared to GCC 10.2.0. 

If we could update the package version again to what was released 2 days ago, PlatformIO would then be able to compile the latest Arduino-ESP8266 core which uses C++17. Currently, when using the very latest `toolchain-xtensa` version available in normally, we get one compilation error / missing include file which exists in the newer version. (See https://github.com/esp8266/Arduino/issues/7792 and https://community.platformio.org/t/how-to-use-c-17-with-esp8266/18950/4?u=maxgerhardt).

mcspr, 2021-01-30:
It is also interesting that the link you provided lists my published package with version mcspr/toolchain-xtensa@5.100200.201223 as well as the platformio's, which was made 3m ago specifically to build the staging with with the original gcc-10.2 pr and then when libc was updated to newlib-4
(...as it is still managed manually by PIO team, might as well do it since I am interested in running the git version)

maxgerhardt, 2021-01-30:
Oh, indeed! They copied it into their registry it seems. The `platformio.ini`

```
[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
; windows specific download link straight from git
platform_packages = 
    toolchain-xtensa@~5.100200.0
    framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
```

works too and actually picks the `mcspr/toolchain-xtensa` option.

```
Processing esp01_1m (platform: espressif8266; board: esp01_1m; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing toolchain-xtensa @ ~5.100200.0
Tool Manager: Warning! More than one package has been found by toolchain-xtensa @ ~5.100200.0 requirements:
 - platformio/toolchain-xtensa @ 2.100100.200706
 - mcspr/toolchain-xtensa @ 5.100200.201223
Tool Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa @ 5.100200.201223 has been installed!
..
================ [SUCCESS] Took 10.54 seconds ================
```

So this also works although it's only the second-last version. 

Probably due to issue https://github.com/platformio/platform-espressif8266/issues/239. The current "stable" version doesn't enforce that the package owner is `platformio` so it will install the non-platformio version. Maybe with that landed it might prefer `platformio/toolchain-xtensa @ 2.100100.200706`. 

Hm... I guess I'll leave this as a note then to update to the very latest option.

mcspr, 2021-01-30:
re. latest version - see https://github.com/platformio/platformio-core/issues/3828

+1 to the https://github.com/platformio/platform-espressif8266/issues/239

commit 326f4392 (2021-05-19): Update Arduino core to the latest v3.0 // Issue #249

Additionally resolves #242
closed 2021-05-19 by valeros