platformio/platform-espressif8266 #244: ESP8266 Flash usage increased (notably) in 2.6.3 compared to 2.6.2

gnalbandian, 2021-02-09:
Hi.
I updated Espressif 8266  platform to 2.6.3, from 2.6.2 and notice a huge increse in flash consumption for the same program.

Compiled with 2.6.2
[image]

Compiled with 2.6.3
[image]

Anyone else have notice this behavior?

gnalbandian, 2021-02-09:
Just as I submitted this, I realize  the issues is not the consumed flash, but the total amount available recognized by the compiler.
In my platformio.ini I don't have defined the flash size. I am using an ESP-07 with 32mbit flash (4M). Somehow, 2.6.2 seemed to picked up the flash size correctly form my definition of "board = esp07", but 2.6.3 failed at this.

Any hint on how to fix this?

gnalbandian, 2021-02-10:
FIXED
I fixed the issue by adding` board_build.ldscript = eagle.flash.4m.ld` at platformio.ini

It seems to me that something have changed at boards definition form 2.6.2 to 2.6.3 that doesn't recognize flash size as it did previusly.

valeros, 2021-02-10:
The reason is the PR https://github.com/platformio/platform-espressif8266/pull/204. According to https://www.esp8266.com/wiki/doku.php?id=esp8266-module-family#summary_table `ESP-07` only has 1MB. In you case you need to use `ESP-07S`.

closed 2021-02-10 by valeros