platformio/platform-espressif8266 #247: Google Cloud Build :- Error: Detected unknown package 'espressif8266'

ben700, 2021-04-20:
[log.txt](https://github.com/platformio/platform-espressif8266/files/6339728/log.txt)

Build ends with

PlatformManager: Installing espressif8266
Error: Detected unknown package 'espressif8266'
ERROR
ERROR: build step 0 "sglahn/platformio-core:3.6.0" failed: step exited with non-zero status: 1

valeros, 2021-04-20:
Hi @ben700 ! I'm not sure this problem has something to do with the `espressif8266` platform itself, you'd better ask the maintainer of the `sglahn/platformio-core` image.

closed 2021-04-20 by valeros

ben700, 2021-04-20:
Can you update with a fix

Fix was to change Python container on Google Cloud Build

like

steps:
- name: python
entrypoint: bash
args:
- '-c'
- |
pip install --upgrade platformio
pio run
dir: .

valeros, 2021-04-20:
It might be a good idea to submit a PR to the repository where that docker image is maintained.

ben700, 2021-04-20:
Google Cloud Support sent me toward you

valeros, 2021-04-20:
Probably you need this repo https://github.com/sglahn/docker-platformio-core