platformio/platform-espressif8266 #249: Update core to 3.0.0

ISO-B, 2021-05-18:
New core version has been released https://github.com/esp8266/Arduino/releases/tag/3.0.0

maxgerhardt, 2021-05-18:
FYI until a core update is released (which I don't think will take very long), see [this topic](https://community.platformio.org/t/esp8266-mmu-increase-heap/21488/3?u=maxgerhardt) for some simple `platformio.ini` options and test to get the new 3.0.0 core and test out some new features (e.g. the +[16Kbyte IRAM heap feature](https://github.com/esp8266/Arduino/pull/7060))

maxgerhardt, 2021-05-19:
And what did I say? 4 minutes ago changes were pushed to support the latest 3.0.0 core. Waiting for release ^_^

valeros, 2021-05-19:
Hi guys, the Arduino and toolchain packages have been updated to the latest 3.0 and 10.2.0 accordingly. Please retest your projects with the upstream version:
```ini
[env:d1]
platform = https://github.com/platformio/platform-espressif8266.git
framework = arduino
board = d1
```

sblantipodi, 2021-05-19:
I love those guys! ehm, not in that sense. 😄

maxgerhardt, 2021-05-19:
Works flawlessly for me! I also noted that you already extended the [documentation](https://github.com/platformio/platformio-docs/commit/459f01352ae3eea3b5adfab9ff389f45768fc4ce) with the new MMU features for increased heap, so that it will be displayed at https://docs.platformio.org/en/latest/platforms/espressif8266.html soon, great.

Testing my example linked above with the new platform 

```ini
[env:d1]
platform = https://github.com/platformio/platform-espressif8266.git
framework = arduino
board = d1
build_flags = -D PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED
monitor_speed = 74880
```

Works exactly as before

```
IRAM free:  21592 bytes
DRAM free:  52336 bytes
```

sblantipodi, 2021-05-19:
@valeros but is there some command to force the update?
I'm using PlatformIO under CLion IDE and it still use 2.6.3

maxgerhardt, 2021-05-19:
Remove `C:\Users\<user>\.platformio\platforms\espressif8266@src-xxxxx` if the folder exists, to force a redownload. You may have previously used a development version.

sblantipodi, 2021-05-19:
@maxgerhardt done it but it redownload the old version:
```

====================[ Build | Production | glowwormluciferinfull_esp8266 ]======
C:\Users\sblantipodi\AppData\Local\JetBrains\Toolbox\apps\CLion\ch-0\211.6693.114\bin\cmake\win\bin\cmake.exe --build C:\dev\dpsoftware\ArduinoRaspbery\PUBLIC\glow_worm_luciferin\cmake-build-glowwormluciferinfull_esp8266 --target Production -- -j 9
Processing glowwormluciferinfull_esp8266 (platform: espressif8266; board: d1_mini_lite; framework: arduino)
--------------------------------------------------------------------------------
Platform Manager: Installing espressif8266
Platform Manager: espressif8266 @ 2.6.3 has been installed!
The platform 'espressif8266' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Verbose mode can be enabled via `-v, --verbose` option
```

sblantipodi, 2021-05-19:
but if you go here:
https://github.com/platformio/platform-espressif8266/releases

there is no new release O_o

sblantipodi, 2021-05-19:
edit: I am trying core 3.0 on my projects, it broke most of the libs I use, most of them are very popular...
it seems that core 3.0 will not be an easy transition...

maxgerhardt, 2021-05-19:
>how those things works? will we see a release soon?

They'll probably release it as a new stable new version after it has been tested for some while / no issue reported.

>platform: espressif8266

If you want to test it you can't use `platform = espressif8266` right now but `platform = https://github.com/platformio/platform-espressif8266.git` as @valeros said above.

sblantipodi, 2021-05-19:
> Which ones? Do the people at the Arduino core already know about this?

FastLED is the most broken one

maxgerhardt, 2021-05-19:
Yeah but per https://github.com/FastLED/FastLED/issues/1236 and https://github.com/FastLED/FastLED/issues/1237 they seem to have been informed about this.

sblantipodi, 2021-05-19:
one of that issue is mine :)

sblantipodi, 2021-05-20:
at this point how can I revert to a previous release if they release the 3.0?
is it possible via the platformio.ini?

sblantipodi, 2021-05-30:
3.0 broke so much things, I hope you'll not release it that soon :D

valeros, 2021-06-03:
Updated in the latest release https://github.com/platformio/platform-espressif8266/releases/tag/v3.0.0

closed 2021-06-03 by valeros