platformio/platform-espressif8266 #252: PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED unreliable if used in a build

TD-er, 2021-06-23:
I have been playing around with the new esp8266/Arduino 3.0.0 and its new option to create a second heap in 16k IRAM

Using `build_flags = -DMMU_IRAM_SIZE=0xC000 -DMMU_ICACHE_SIZE=0x4000 -DMMU_IRAM_HEAP` is very unreliable with regards to whether it is compiled as such.
Using the define `PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED` is already a bit more clear if it is being used or not.

But still it is rather hit or miss if the feature is active or not.

I now use this as part of my PIO configuration:
```
build_flags               = ${esp82xx_3_0_x.build_flags}
                            -DPIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED 
                            -DUSE_SECOND_HEAP
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK3
                            -Wno-deprecated-declarations
```
The define `USE_SECOND_HEAP` is used in my code to wrap the calls to switch heap, like this:
```c++
    case LabelType::FREE_MEM:               return String(ESP.getFreeHeap());
#ifdef USE_SECOND_HEAP
    case LabelType::FREE_HEAP_IRAM:
      {
        HeapSelectIram ephemeral;
        return String(ESP.getFreeHeap());
      }
#endif
```

It is also used for an enum value and some code to actually get the free heap for DRAM or IRAM.
But with the same code, only a recompile, I do get it to actually show the different free heap values and in a next build it just shows the same value for both given I call above code for both enums.

Just as an illustration a tab I had still open from this morning and a tak of a more recent build running on the same node:
[image]

Earlier build:

<html>
<body>

Free RAM: | 10872 (8688 - sendContentBlocking)
-- | --
Free Heap IRAM: | 17552

</body>
</html>

Later build:
<html>
<body>

Free RAM: | 12288 (10104 - sendContentBlocking)
-- | --
Free Heap IRAM: | 12288

</body>
</html>

valeros, 2021-06-25:
Hi @TD-er ! The MMU configuration is handled by [this code](https://github.com/esp8266/Arduino/blob/master/tools/platformio-build.py#L287-L341). `PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED` macro adds the following defines to the compile command:
```
-D MMU_IRAM_SIZE=0xC000
-D MMU_ICACHE_SIZE=0x4000
-D MMU_IRAM_HEAP
```
It's a pretty straightforward configuration, so TBH I'm not sure why it might be unreliable. Are there any changes between two consecutive compilations?

TD-er, 2021-06-25:
I got the idea the building process on its self might be flaky.
It doesn't matter if I use the PIO define or the ones you mention (with the `-DPIO_FRAMEWORK_ARDUINO_MMU_CUSTOM` or else you get lots of complaints about redefines)

Just some observations:
- Make a build -> OK, 2nd heap is present
- Do some tests, decide things don't work out, thus revert to previous state of code
- Make a build -> sometimes OK, sometimes 2nd heap is not present.

N.B. I don't build other PIO envs inbetween and don't touch the PIO ini files.

What does happen every now and then -now that I think of it- is that my PIO project tasks tree gets folded with an infinite waiting state (like what you have when unfolding it for the first time, showing "Loading...") unless you close the item and open it again.

I mean like pressing here:
[image]

This does seem to happen more often when you toggle between tasks like "build", "clean" and "Upload and Monitor".
Could this also affect the build process?

valeros, 2021-06-25:
> Could this also affect the build process?

Highly unlikely. That task tree simply runs the `pio project data` command on your project to retrieve the build information. So if you manually run `pio project data --json-output` command in the IDE terminal you will see all the build flags and targets available (including `PIO_FRAMEWORK_ARDUINO_MMU_CACHE16_IRAM48_SECHEAP_SHARED`).

In any case, your freshly edited files will trigger linker to reassemble the final elf image and there might something wrong with autogenerated linker scripts.

Anyway, could you please prepare a minimal example so I can reproduce it on my machine?

TD-er, 2021-06-25:
Well a "minimal" example is quite hard as I don't have a clue here what triggers it.

By the way, the IDE just suddenly "folded" the project tasks as I described, which happens every now and then.
So as you said, not likely to be causing this problem, but more to indicate what happens as I mentioned it.
[image]

TD-er, 2021-06-25:
I do have the feeling (I know, lots of 'feeling', not yet hard facts) that it may somehow be related to this issue:
https://github.com/esp8266/Arduino/issues/8169

The reason I got the feeling it may be related is that I'm having a number of issues when running the current staging code of esp8266/Arduino, regardless if it is using SDK3 or SDK222 and also with or without experimenting with the 2nd heap.

What I'm seeing (so leave out the idea of the MMU defines) is that the ESP may crash on seemingly unrelated code.
Code that was not changed in a long time.
It does look like code called with flash strings does appear more often on the decoded stack trace, but not always.
Also the WDT resets do occur quite often.

So this does seem like there is something fishy in the way how code is being linked. The order in which it is linked may lead to working or crashing build.
For example I now had it crash with the stack trace in the linked issue. Did a complete clean of the .pio directory, the packages, cache and some other folder in the .platformio directory and did a clean rebuild. Now it is running fine.
With the crashing build it was a rebuild with only a minor change in 1 file and the build before was based on a clean build just like the one running on my node.

I hope this is enough defense to give my 'gut feeling' some credits ;)

TD-er, 2021-06-25:
I have another idea of what may be happening here...

In VS code, you have on the right the various PIO tasks once done so you can switch between them.

[image]

What I now noticed was that I was building a build before that did not have the 2nd heap stuff defined for the PIO env.
So I cleared all build files etc. and selected from another PIO env. to "upload and monitor". (an env which should have it defined)
It starts building, uploading and again no 2nd heap in that build.
Then I noticed it built the wrong environment even though I am 100% sure I clicked the right one.

Clicked this one:
[image]

But since a previous task with "Upload and Monitor" from another env was still present in this list in the lower right corner of VS code, it built that one instead.

So it looks like the re-use of tasks is picking the wrong environment.

TD-er, 2021-06-25:
Ah proof!

What is pressed:
[image]

What is executed: 
[image]

[image]

TD-er, 2021-06-25:
And it is very well reproducible.
At least when I "go up" in the project tasks.

[image]
Had "custom_beta_ESP8266_4M1M" via Upload and Monitor and then selected "custom_ESP8266_4M1M" Upload and monitor, but it selected the previous one, as long as the task was still present in the lower right corner of VS code.

After closing those tasks there, the actual clicked task was executed.

valeros, 2021-06-30:
So it seems the issue has nothing to do with the `PIO_FRAMEWORK_ARDUINO_MMU` options as it's rather a PlatformIO for VSCode plugin bug. Does it make sense to move this issue to the https://github.com/platformio/platformio-vscode-ide repository?

TD-er, 2021-06-30:
Yep, I guess so.