platformio/platform-espressif8266 #255: Update to Arduino Core 3.0.2

mcspr, 2021-07-27:
New version has been released https://github.com/esp8266/Arduino/releases/tag/3.0.2

closed 2021-07-27 by valeros

valeros, 2021-07-27:
Updated, please retest using the upstream version of the platform:
```ini
[env:nodemcuv2]
platform = https://github.com/platformio/platform-espressif8266.git
framework = arduino
board = nodemcuv2
```

mcspr, 2021-07-27:
It should also include toolchain bump to resolve inttypes.h issue, can't build the mcve from the linked issue
https://github.com/esp8266/Arduino/issues/8220
https://github.com/esp8266/Arduino/pull/8222

valeros, 2021-07-28:
Thanks, toolchain packages have been updated as well, please run `pio platform update espressif8266` in your terminal.

mcspr, 2021-07-28:
it works, thanks!

mcspr, 2021-08-12:
@valeros should this be tagged for platform release?

commit 10c64d02 (2021-07-27): Update Arduino core to v3.0.2

Resolve #255