platformio/platform-espressif8266 #259: New PlatformIO core with old platform-espressif8266 makes platform tasks disappear

maxgerhardt, 2021-09-23:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Win 10 x64

**PlatformIO Version** (`platformio --version`): 5.2.1b2

### Description of problem

Using an ESP8266 project which specifies the usage of an older platform-espressif8266 version makes the "Platform" tasks disappear (e.g., "Upload Filesystem Image"), although that platform version is capable of doing that.

The PlatformIO core APIs (or the VSCode plugin) should still work when an older platform version is selected for backwards compatibility reasons. Many projects exists that use previous platform versions, and users not seeing the "Upload Filesystem" button is confusing for them.

#### Steps to Reproduce

1. Clone https://github.com/maxgerhardt/pio-old-esp8266/tree/old_platform in the `old_platform` branch
2. Import into VSCode + PlatformIO
3. Observe available tasks

### Actual Results

No "platform" tasks

[image]

### Expected Results

Platform tasks present as when using the latest platform version.

[image]

### If problems with PlatformIO Build System:

Per referenced repo. Critical line is

```ini
platform = espressif8266@3.2.0
```

### Additional info

valeros, 2021-09-23:
Hi @maxgerhardt ! Thanks for reporting. It's strange, but I'm not able to reproduce the problem. I cloned that project example, opened in VSCode and I can see the platform-specific tasks:
[image]

Maybe it's a typo, but in the example you use v3.2.0 which is currently the latest available, so I went further and downgraded the platform to v3.0.0 and again the tasks were displayed just fine:
[image]

Any chance there might be something wrong with your platform installations?

maxgerhardt, 2021-09-23:
Are you 100% sure you checked out the `old_platform` branch of the project? It's using 2.2.0

https://github.com/maxgerhardt/pio-old-esp8266/blob/9c3713fcad0179472daadfdc98663ccf04991903/platformio.ini#L37-L41

ivankravets, 2021-09-23:
The oldest versions do not use [env.AddPlatformTarget](https://github.com/platformio/platform-espressif8266/blob/develop/builder/main.py#L359) API. Do we have a lot of requests for the oldest versions? The simplest solution is to call target directly: `pio run -t uploadfs`, etc.

maxgerhardt, 2021-09-23:
That is also the workaround I proposed in the thread. While that works, many users may be confused why the upload task isn't there in the project and just give up. I don't have a number on that though. The project in question where this occurs is a 3D printer which **at this time** still uses the old platform: https://github.com/luc-github/ESP3D/blob/39bfde2586a55e5327f00fc04a92926469bc5853/platformio.ini#L39-L41