platformio/platform-espressif8266 #262: How to make platformio compile header file in the project first？

ethereal14, 2021-11-13:
I want to use i2c on esp8266. First I copy the `i2c_master.h` and `i2c_master.c` into my project. I want use `GPIO12` and `GPIO14` as i2c_sda and i2c_scl . Then I change it is macro definition

original

```c
#define I2C_MASTER_SDA_MUX PERIPHS_IO_MUX_GPIO2_U
#define I2C_MASTER_SCL_MUX PERIPHS_IO_MUX_MTMS_U
#define I2C_MASTER_SDA_GPIO 2
#define I2C_MASTER_SCL_GPIO 14
#define I2C_MASTER_SDA_FUNC FUNC_GPIO2
#define I2C_MASTER_SCL_FUNC FUNC_GPIO14
```

now

```c
#define I2C_MASTER_SDA_MUX PERIPHS_IO_MUX_MTDI_U
#define I2C_MASTER_SCL_MUX PERIPHS_IO_MUX_MTMS_U
#define I2C_MASTER_SDA_GPIO 12
#define I2C_MASTER_SCL_GPIO 14
#define I2C_MASTER_SDA_FUNC FUNC_GPIO12
#define I2C_MASTER_SCL_FUNC FUNC_GPIO14
```

But when I compile this project and upload firmware to esp8266,  it is i2c pin still original pin. 

I know it is usesing `i2c_master.h`  from the SDK,but I don't known how to solve this problem

valeros, 2021-11-15:
Hi @ethereal14 ! This repository is not the right place to ask such generic questions. Next time please ask for help here https://community.platformio.org.  In a nutshell, although I'm not sure what framework/library you're using, but based on your code snippet, you can try to redefine the I2C pins globally using the [build_flags](https://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags) option.

closed 2021-11-15 by valeros

ethereal14, 2021-11-15:
Thanks for your reply!@valeros , you gave me a good idea!🙏