platformio/platform-espressif8266 #263: Upload Filesystem Image OTA (uploadfsota) fails to provide -s flag to espota

cregganna, 2021-11-23:
Hi,

I have found an issue when uploading Filesystems via OTA. The "-s" flag is not supplied to the espota call and thus the OTA fails.

The issue is at:
https://github.com/platformio/platform-espressif8266/blob/bb547cee8e61815ee33b9ea0b6c03449196dc269/builder/main.py#L311

When running OTA FS updates the Target is `"uploadfsota"` not `"uploadfs"`.

With the original code I see (Note spiffs is reported as False)
```
Uploading .pio/build/esp12f_upload_ota/littlefs.bin
21:45:16 [DEBUG]: Options: {'esp_ip': '192.168.101.39', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 52607, 'auth': 'XXXXXXXX', 'image': '.pio/build/esp12f_upload_ota/littlefs.bin', 'spiffs': False, 'debug': True, 'progress': True}
21:45:16 [INFO]: Starting on 0.0.0.0:52607
21:45:16 [INFO]: Upload size: 1024000
21:45:16 [INFO]: Sending invitation to: 192.168.101.39
Authenticating...OK
21:45:17 [INFO]: Waiting for device...

Uploading: [                                                            ] 0% 
....
Uploading: [============================================================] 100% Done...

21:45:32 [INFO]: Waiting for result...

21:45:32 [ERROR]: ERROR[10]: 
*** [uploadfsota] Error 1
```

With the code modified to:
```
if "uploadfsota" in COMMAND_LINE_TARGETS:
```

I see (Note spiffs is reported as True)
```
Uploading .pio/build/esp12f_upload_ota/littlefs.bin
21:48:48 [DEBUG]: Options: {'esp_ip': '192.168.101.39', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 20550, 'auth': 'XXXXXXXX', 'image': '.pio/build/esp12f_upload_ota/littlefs.bin', 'spiffs': True, 'debug': True, 'progress': True}
21:48:48 [INFO]: Starting on 0.0.0.0:20550
21:48:48 [INFO]: Upload size: 1024000
21:48:48 [INFO]: Sending invitation to: 192.168.101.39
Authenticating...OK
21:48:49 [INFO]: Waiting for device...

Uploading: [                                                            ] 0% 
...
Uploading: [============================================================] 100% Done...

21:49:04 [INFO]: Waiting for result...
21:49:05 [INFO]: Result: OK
```

And the new FS is now loaded on the device.

With the change the non-OTA FS updates are still functional.

becks0815, 2022-01-31:
I can confirm this. Tried to upload a program which uses littlefs via ota. Got the same response
Uploading: [============================================================] 100% Done...

22:36:48 [ERROR]: ERROR[10]:
*** [uploadfsota] Error 1

After changing main.py like mentioned above the error was gone.

theplen, 2022-03-28:
Agreed, 

https://github.com/platformio/platform-espressif8266/blob/bb547cee8e61815ee33b9ea0b6c03449196dc269/builder/main.py#L311

Changed to:
`if "uploadfsota" in COMMAND_LINE_TARGETS:`

Fixes the "Upload Filesystem Image OTA" issue.

crazyduck, 2022-06-13:
Working also for me.
How can this change be integrated in latest ESP8266 pio files?

Helix70, 2023-05-29:
This fixed it for me too. Not sure why this is not being adopted into the code, it is pretty clear.

lycheejp, 2023-08-05:
Super merci beaucoup cela fonctionne parfaitement de mon coté

itldg, 2023-10-31:
This solved my problem, thanks

PR #314 (2024-01-22): Fix issues #263
fixes #263
commit 6d98803d (2024-01-22): Add missing upload flag for uploadfsota target (#314)

Fixes #263
closed 2024-01-22 by valeros