platformio/platform-espressif8266 #264: Correct Setting For Shelly Board

GamerClassN7, 2021-12-29:
Hello i struggle with corect board setting for Shelly 1 Boards 

ewerithing i foung
force Board to random restarts
EEprom not work or OTA not work 

did anyone have corret and verified setting fro Shelly Boards ?

valeros, 2022-01-03:
Hi @GamerClassN7 ! I don't see any boards named `Shelly` in the [list of officially support hardware](https://registry.platformio.org/packages/platforms/platformio/espressif8266/boards). Please forward your issue to the source where you found instructions for this board.

closed 2022-01-03 by valeros

thedanbob, 2023-01-10:
For what it’s worth, I got the Shelly 1 working in my project with this config:

```json
{
  "build": {
    "arduino": {
      "ldscript": "eagle.flash.2m.ld"
    },
    "core": "esp8266",
    "extra_flags": "-DESP8266 -DARDUINO_ARCH_ESP8266 -DLED_BUILTIN=0",
    "f_cpu": "80000000L",
    "f_flash": "40000000L",
    "flash_mode": "qio",
    "mcu": "esp8266",
    "variant": "generic"
  },
  "connectivity": [
    "wifi"
  ],
  "frameworks": [
    "arduino",
    "esp8266-rtos-sdk",
    "esp8266-nonos-sdk"
  ],
  "name": "Shelly 1",
  "upload": {
    "maximum_ram_size": 81920,
    "maximum_size": 2097152,
    "require_upload_port": true,
    "resetmethod": "ck",
    "speed": 115200
  },
  "url": "https://shelly.cloud/knowledge-base/devices/shelly-1/",
  "vendor": "Shelly"
}
```

Put that in `boards/shelly1.json` and `board = shelly1` in platformio.ini.