platformio/platform-espressif8266 #271: Wrong PWM resolution

FernandoGarcia, 2022-03-22:
Hi!

In the versions above 2.6.3 the `analogWrite` function works with 8 bit resolution instead 10 bit.

Using the code below you will get 3.3 V instead 1.61 V on output.

```cpp
#include "Arduino.h"

const int LED_PIN = D7;
const int PWM_VALUE = 500;

void setup()
{
  pinMode(LED_PIN, OUTPUT);
  analogWrite(LED_PIN, PWM_VALUE);
}

void loop()
{
}
```
Here some pics showing the problem.

Code:

[image]

Platformio.ini using v3.2.0:

[image]

Oscilloscope measurements:

[image]

Platformio.ini using v2.6.3

[image]

Oscilloscope measurements:

[image]

Best regards.

mcspr, 2022-03-22:
See https://github.com/esp8266/Arduino/pull/7456 and https://github.com/esp8266/Arduino/pull/2895
Notice that this repo is for PlatformIO-specific build tools, Arduino Core related issues and it's development are at https://github.com/esp8266/Arduino

FernandoGarcia, 2022-03-22:
Thanks you very much for the info.
Sorry by post the issue in wrong place.

closed 2022-03-22 by FernandoGarcia