platformio/platform-espressif8266 #275: Export CCFLAGS macros to IntelliSense

maxgerhardt, 2022-05-22:


ivankravets, 2022-05-22:
Thanks for the report! We need to fix a build script and put macro to the right scope. 

Regarding `CCFLAGS`. I remember we had a problem with VSCode's C/C++ plugin if we pass GCC flags. So, we don't pass all compiler flags. The only flags related to the C/C++ preprocessed are passed.

maxgerhardt, 2022-05-22:
>So, we don't pass all compiler flags. The only flags related to the C/C++ preprocessed are passed.

But how about transforming known `-D..` flags and putting them in the `defines` section of the `c_cpp_properties.json`? They don't have to be in the compiler arguments section. This fix would be more general and no build script adaptions needed.

ivankravets, 2022-05-28:
An integrator decides on which scopes to put flags. We don't do any manipulation with flags. The correct solution is https://github.com/esp8266/Arduino/commit/c1144c5740406693808e92f072872260c7f069a0 because it also fixes ASM builds.

Thanks for the PR!

closed 2022-05-28 by ivankravets