platformio/platform-espressif8266 #279: How to set esp8266 flash layout in vscode + platformIO ?

Qiu-zx, 2022-07-25:
### I couldn't find the answer online.But here are a esp8266 description website [https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html](url)

In the website,I can't understand these sentences.

[image]

How to use the particular configuration `FLASH_MAP_SETUP_CONFIG(FLASH_MAP_OTA_FS)` ?
### I will appreciate if anyone can help me.

mcspr, 2022-07-25:
Notice that you are looking at `latest` aka development documentation, there's no such thing yet in the stable release.

Check out
https://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

And select our generic .ld script (`eagle.flash.auto.ld`). Then, add the macro in your project, so it is visible in some .o and can be picked up by the linker.
https://docs.platformio.org/en/latest/platforms/espressif8266.html#flash-size
https://github.com/esp8266/Arduino/blob/master/tools/sdk/ld/eagle.flash.auto.ld

Also, add `-DFLASH_MAP_SUPPORT=1` to `build_flags = `
https://github.com/esp8266/Arduino/blob/e149829802f351029888c42ab5ce12c0353c42ad/boards.txt#L489

Note that it may change in the future to something a bit more coherent, or in case we somehow adapt pseudo-partition scheme from NONOS v3

Qiu-zx, 2022-07-26:
> Notice that you are looking at `latest` aka development documentation, there's no such thing yet in the stable release.
>
> Check out [...]

### You are my savior,Thanks a lot !

closed 2022-07-26 by Qiu-zx