platformio/platform-espressif8266 #281: D1 mini Pro crashes after executing setup()

LONGACAT, 2022-09-28:
Using PlatformIO to compile for the WEMOS D1 mini Pro will cause a crash after reaching end of setup(). The following sketch works fine in Arduino IDE but not in PlatformIO CLion:
```cpp
#include <Arduino.h>

void setup() {
    Serial.begin(115200);
    Serial.println("Setup started");
    pinMode(LED_BUILTIN, OUTPUT);
    Serial.println("Setup completed");
}

void loop() {
    digitalWrite(LED_BUILTIN, HIGH);
    delay(1000);
    digitalWrite(LED_BUILTIN, LOW);
    delay(1000);
}
```

Expected behaviour (this happens in Arduino IDE):
Prints "Setup started" and "Setup completed" to Serial monitor then blinks builtin led

Actual behaviour: 
Prints "Setup started" and "Setup completed" to Serial monitor then resets with cause 2

platformio.ini:

```ini
[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino
monitor_speed = 115200
```

After adding `build_type = debug` and `monitor_filters = esp8266_exception_decoder` I have gotten the following from Serial monitor (it gets repeated infinitely):
```
Setup started
Setup completed

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40201060 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000010 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x40201060 in setup at ??:?

>>>stack>>>

ctx: cont
sp: 3ffffe00 end: 3fffffc0 offset: 0190
3fffff90:  3fffdad0 00000000 3ffee4e8 40201060  
3fffffa0:  feefeffe feefeffe 3ffe85d8 40201af7  
3fffffb0:  feefeffe feefeffe feefeffe 40100d61  
<<<stack<<<

0x40201060 in setup at ??:?
0x40201af7 in loop_wrapper() at core_esp8266_main.cpp:?
0x40100d61 in cont_wrapper at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/cont.S:81

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00042630
~ld
```
According to some googling this means the heap is full and the device ran out of memory, but I don't see how is that possible with a simple blink sketch and why does it work fine in Arduino IDE

LONGACAT, 2022-09-28:
After some further testing it seems the issue was introduced in platform version 3.2.0. Using 3.1.0 and below works fine

raphael303, 2022-12-18:
Same problem here. D1 mini pro ESP8266. I have not change anything in the initial yaml:

`esphome:
  name: espd1test

esp8266:
  board: d1_mini_pro

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "WFVROpHyI6GF1gxpjPn50g6LHK+nhf2flOmYZHJp+Y0="

ota:
  password: "0a49bb5de73a59513b052dbbce24022a"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Espd1Test Fallback Hotspot"
    password: "VWomHeBo0tEE"

captive_portal:
    `

I get this in the log on esphome in google chrome.:

`r$l����$�|�l�#|�����|b�b��on�$oN���cp��l{lslp�n��l��bn�|���#�|~�o�l��$`�Nn�s$N{���osr�`p�o�r��ܜ��"N�|�c��on��$`�no�{lnr���n{l r��n{l �䎐�܎l`��N�l[I][logger:258]: Log initialized
[C][ota:469]: There have been 3 suspected unsuccessful boot attempts.
[I][app:029]: Running through setup()...
[C][wifi:037]: Setting up WiFi...
[C][wifi:038]:   Local MAC: 48:3F:DA:60:18:03
[D][wifi:386]: Starting scan...

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00074f70
~ld`