platformio/platform-espressif8266 #286: Error: pio run -t nobuild -t upload

Bighoneypot, 2022-11-25:


Daiki51, 2022-12-06:
I am also encountering the same error.
I would like to add that the same thing happens when the host/remote is Ubuntu 22.

mcspr, 2022-12-06:
Also looks like https://github.com/platformio/platform-espressif32/issues/861

We try to call `env.GetActualLDScript()` to retrieve our flash size info
https://github.com/platformio/platformio-core/blob/48655ad728c25c6af67da913c36651e27fdd05ad/platformio/builder/tools/piomisc.py#L42-L84

Which is broken in nobuild case, since there is no env.BuildProgram() and LINKFLAGS is empty. As a workaround, one could do the same thing as suggested in the esp32 issue - populate it manually with the `-Wl,-T...ld` entry from the currently used board .json
https://github.com/platformio/platform-espressif8266/blob/46d25a4f41791cce1b9269da99b7e71e05aa6ba4/builder/main.py#L86
https://github.com/platformio/platform-espressif8266/blob/46d25a4f41791cce1b9269da99b7e71e05aa6ba4/builder/main.py#L93
https://github.com/platformio/platform-espressif8266/blob/46d25a4f41791cce1b9269da99b7e71e05aa6ba4/builder/main.py#L124

Not really sure if nobuild is implemented correctly, though. Perhaps it should still invoke pre-requisite actions but somehow inhibit the PROGRAM .bin regeneration when scons thinks any of the targets need to be built i.e. somehow tell it that all of our dependencies for .bin are up-to-date

Jason2866, 2023-02-27:
This one worked in my test for `pio run -t nobuild -t upload`
```
#
# The script sets the missing "LDSCRIPT_PATH" when using the command `pio run -t nobuild`
# Adopted from https://github.com/platformio/platform-espressif32/issues/861#issuecomment-1241871437
# Possible now to upload the firmware or the filesystem with (when builded already!):
#
# `pio run -t nobuild -t upload` and `pio run -t nobuild -t uploadfs`
#

Import("env")

import os
from SCons.Script import COMMAND_LINE_TARGETS

board_config = env.BoardConfig()
framework_dir = env.PioPlatform().get_package_dir("framework-arduinoespressif8266")
assert os.path.isdir(framework_dir)

if "nobuild" in COMMAND_LINE_TARGETS:
    env.Replace(
        LDSCRIPT_PATH=os.path.join(
            framework_dir,
            "tools",
            "sdk",
            "ld",
            board_config.get("build.arduino.ldscript"),
        )
    )
#    print("Set LDSCRIPT_PATH to: ", os.path.join(framework_dir,"tools","sdk","ld",board_config.get("build.arduino.ldscript")))
```