platformio/platform-espressif8266 #292: Clang as the compiler?

MacDada, 2023-02-17:
Would it be possible to use Clang instead of GCC to compile for esp8266 board?

I've already [made it working for `native` platform](https://github.com/platformio/platform-native/issues/22#issuecomment-1435140844), but same config/script for `espressif8266` gives errors below:

```
➜  DnWiFiDoorLock git:(master) ✗ pio run -e nodemcu   
Processing nodemcu (platform: espressif8266; board: nodemcuv2; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing me-no-dev/ESP Async WebServer
Unpacking  [####################################]  100%
Library Manager: ESP Async WebServer@1.2.3 has been installed!
Library Manager: Resolving dependencies...
Library Manager: Installing ESPAsyncTCP
Unpacking  [####################################]  100%
Library Manager: ESPAsyncTCP@1.2.2 has been installed!
Library Manager: Installing ayushsharma82/WebSerial @ ^1.3.0
Unpacking  [####################################]  100%
Library Manager: WebSerial@1.3.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (4.0.1) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 48 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3
|   |-- ESPAsyncTCP @ 1.2.2
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0
|-- WebSerial @ 1.3.0
|   |-- ESP8266WiFi @ 1.0
|   |-- ESPAsyncTCP @ 1.2.2
|   |-- ESP Async WebServer @ 1.2.3
|   |   |-- ESPAsyncTCP @ 1.2.2
|   |   |-- Hash @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
|-- DnWiFiDoorLock
|   |-- DnAppArduino
|   |   |-- DnAppHardware
|   |   |-- DnAppLogger
|   |   |   |-- DnAppCommon
|   |   |   |   |-- cstdarg
|   |   |-- DnAppCommon
|   |   |   |-- cstdarg
|   |-- DnAppEsp
|   |   |-- DnAppCommon
|   |   |   |-- cstdarg
|   |   |-- DnAppLogger
|   |   |   |-- DnAppCommon
|   |   |   |   |-- cstdarg
|   |   |-- WebSerial @ 1.3.0
|   |   |   |-- ESP8266WiFi @ 1.0
|   |   |   |-- ESPAsyncTCP @ 1.2.2
|   |   |   |-- ESP Async WebServer @ 1.2.3
|   |   |   |   |-- ESPAsyncTCP @ 1.2.2
|   |   |   |   |-- Hash @ 1.0
|   |   |   |   |-- ESP8266WiFi @ 1.0
|   |-- DnAppHardware
|   |-- DnAppLogger
|   |   |-- DnAppCommon
|   |   |   |-- cstdarg
|   |-- ESP Async WebServer @ 1.2.3
|   |   |-- ESPAsyncTCP @ 1.2.2
|   |   |-- Hash @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
|   |-- Servo @ 1.0.2
|   |-- WebSerial @ 1.3.0
|   |   |-- ESP8266WiFi @ 1.0
|   |   |-- ESPAsyncTCP @ 1.2.2
|   |   |-- ESP Async WebServer @ 1.2.3
|   |   |   |-- ESPAsyncTCP @ 1.2.2
|   |   |   |-- Hash @ 1.0
|   |   |   |-- ESP8266WiFi @ 1.0
|   |-- DnAppCommon
|   |   |-- cstdarg
|   |-- ESP8266mDNS @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- ArduinoOTA @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
|   |   |-- ESP8266mDNS @ 1.2
|   |   |   |-- ESP8266WiFi @ 1.0
Building in release mode

Replacing the default toolchain with Clang…

Default CC (env): xtensa-lx106-elf-gcc
Default CXX (env): xtensa-lx106-elf-g++

Replaced CC (env): clang
Replaced CXX (env): clang++

Default CC (projenv): xtensa-lx106-elf-gcc
Default CXX (projenv): xtensa-lx106-elf-g++

Replaced CC (projenv): clang
Replaced CXX (projenv): clang++

Compiling .pio/build/nodemcu/src/arduino.cpp.o
Generating LD script .pio/build/nodemcu/ld/local.eagle.app.v6.common.ld
Compiling .pio/build/nodemcu/libc46/ESPAsyncTCP/AsyncPrinter.cpp.o
Compiling .pio/build/nodemcu/libc46/ESPAsyncTCP/ESPAsyncTCP.cpp.o

clang-15: error: unknown argument '-mlongcalls'; did you mean '-mlong-calls'?
clang-15: error: unknown argument: '-mtext-section-literals'
clang-15: error: unknown argument '-free'; did you mean '-fsee'?
clang-15: error: unknown argument: '-fipa-pta'

*** [.pio/build/nodemcu/src/arduino.cpp.o] Error 1
```

valeros, 2023-03-21:
Hi @MacDada ! Unfortunately, it's currently not possible to use Clang with esp8266.

MacDada, 2023-03-24:
https://github.com/espressif/arduino-esp32/issues/6858

[image]