platformio/platform-espressif8266 #298: Arduino + ESP8266-RTOS-SDK

felipeapp, 2023-05-27:
Is that possible to use Arduino platform and RTOS-SDK? I have a project using arduino platform and I would like to use the mqtt_client.h library from RTOS-SDK. Any tips? Thanks!

cziter15, 2024-04-10:
It will be cool to get even rtos alone ;)

The one which is hosted on the registry is pretty ancient 1.5.0, while there's 3.4 out there.

As far as I know there was a plan to move to RTOS @ arduino-esp8266, but there are some road blocks, so I wouldn't expect it to happen.

See:
https://github.com/esp8266/Arduino/issues/5790

modem-man-gmx, 2024-09-25:
As far as I can see, my setup is using either RTOS 3.4 pure or, hidden behind Arduino, a RTOS 3.2 core, since some years.
I installed Platform "Espressif 8266" in Version 4.1.0.
Did you accidentally stuck with one of the two other offered 8266 platforms?

[image]