platformio/platform-espressif8266 #302: Screen gets stuck at the 'Please Wait' window

chomchom69420, 2023-08-04:
What kind of issue is this?

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 11 

**PlatformIO Version** (`platformio --version`): 6.1.9

### Description of problem
The screen gets stuck at the 'Please Wait' window. On clicking somewhere else, the 'Please Wait' window disappears and the project can be seen to be created.  

#### Steps to Reproduce

1. Create a project on esp32 wroom devkit (not sure if this will reproduce the issue)

### Actual Results
[image]

The main.cpp file also doesn't show up in the ./src folder

I tried manually creating the main.cpp file and adding some code. It does not recognize the Arduino.h file when I do a #include.  Neither does it recognize the Wifi.h file. These are files that were always recognized by default. 

### Expected Results
The window should close and the project tab should open up. 

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:esp_wroom_02]
platform = espressif8266
board = esp_wroom_02
framework = arduino
```

basprins, 2025-02-14:
for those who encouter the same problem, just helped a friend with the same problem.
on windows, you can use process explorer to figure out which command is running when the dialog is open. 
when you execute that command inside a vscode terminal, after activating the venv
C:\Users\<username>\.platformio\scripts\penv
.\Activate.ps1

then copy paste the command that you see in process explorer

then go one directory up to be in C:\Users\<username>\.platformio\scripts\
and manually run
.\platformio.exe init --board esp32doit-devkit-v1 --sample-code --project-option "framework = espidf" --ide vscode

then you will probably see that the scripts starts to run, and unpacks some toolchains and installs some debuggers. 

On his hardward (1ghz cpu, 8gb ram) this took about 50 minutes. So in his case, it wasn't stuck, it just took... really... long.

Maybe a nice option is to add command line output to this project wizard, so that users can see that something is still going on.

Strat00s, 2025-03-21:
Almost 2 years later and there is still no useful output about what is going on. This is just plain bad UX.