platformio/platform-espressif8266 #324: Is it possible to specify the ArduinoOTA password in PlatformIO command line?

mifarelight, 2025-05-08:
I am used to uploading firmware to ESP32 via PlatformIO command line by using ArduinoOTA, such as:
`pio run -t upload --upload-port 192.168.1.128`

I add these lines to platformio.ini:
```
upload_flags =
	--auth=blablablabla
```

The upper command works fine.

I'm wondering if there is a cli way to do this without having to modify the platformio.ini file, like:
`pio run -t upload --upload-port 192.168.1.128 -a blablablabla`

valeros, 2025-05-09:
Hi @mifarelight, there is no easy way to do that, but you can try get by with the [PLATFORMIO_UPLOAD_FLAGS](https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_UPLOAD_FLAGS) environment variable.

mifarelight, 2025-05-11:
> Hi [@mifarelight](https://github.com/mifarelight), there is no easy way to do that, but you can try get by with the [PLATFORMIO_UPLOAD_FLAGS](https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_UPLOAD_FLAGS) environment variable.

Thanks for your reply.
The solution you mentioned is what I have used.
The source of my question is: if I unfortunately use different authentication passwords on more than one hardware, and now I want to upgrade them all to the same firmware, I have to recompile the entire project for each different "--auth=xxxxxxxxxxxxxxxx".

At least, I know that I should choose other firmware update methods instead of ArduinoOTA.

closed 2025-05-11 by mifarelight